jlecomte/ascom-wireless-flat-panel #1: PCBWay information

alexd106, 2024-01-14:
Hi, First, thanks for putting this excellent project together. I would like to suggest an enhancement to the documentation regarding information required for PCB manufacturer for those of us who have never used this service before. Before uploading the gerber files to PCBWay, the user will need to specify a number of options like PCB dimensions, material, min hole spacing etc. Possibly these are all default values (except dimensions which I can get from Kicad) but it would be great if you can add some information regrading this to the documenttation. Many thanks.

Wellco, 2024-01-14:
@alexd106 I just uploaded Julien's zip file into the PCBway viewer and then took a few mins to review the design based on resources here in GITHub. Then I clicked order and it looked fine to me, or am I missing something. I have never used PCBway before so I'm not an expert.

alexd106, 2024-01-14:
@Wellco thanks for your response, I haven't used it before either and am probably overcomplicating things! I didn't use the PCBWay viewer (didn't know it existed!) just went to the PCBWay website and started to go through their process (you need to specify basic options before you can upload the zip file this way). Your way seems much more sensible :). I noticed that when you click on the 'get instant quote' in PCBWay viewer none of these options are prefilled in the quote but perhaps that doesn't matter if they have the PCB  file.

Wellco, 2024-01-14:
@alexd106 I don't know if this helps, but go to PCBway (https://www.pcbway.com/project/OnlineGerberViewer.html), then load Julien's zip file. Then have a look or just click "get and instant quote". I think that's all there is to it. Let me know if you think I am wrong.

jlecomte, 2024-01-14:
Addressed in [this commit](https://github.com/jlecomte/ascom-wireless-flat-panel/commit/bf930bba483c1226f2a90652a6875b9ac90d640e). Thanks!

closed 2024-01-14 by jlecomte

alexd106, 2024-01-14:
Hi, noticed your closed issue regarding the non-availability of the Tayola
diffuser in the UK. I'm also in the UK and found this on amazon
https://www.amazon.co.uk/gp/product/B09YRFC3JS/ref=ox_sc_act_title_2?smid=A1NNPUA7EKFNWH&th=1
Haven't tried it yet but have ordered and will see if it fits (or the
freecad model can be modified).
HTH