jlecomte/ascom-wireless-flat-panel #2: I'm not able purchase the Taloya ceiling light.

Wellco, 2024-01-14:
I'm in the UK and can't get the Taloya ceiling light; Is it possible to document some basic specs for a generic LED strip?  for instance would this be an option 

https://www.aliexpress.com/item/1005004714205956.html?spm=a2g0o.productlist.main.3.10aa5438pM62a3&algo_pvid=fd07f7f2-526f-4abe-bfac-e396d3190bf2&algo_exp_id=fd07f7f2-526f-4abe-bfac-e396d3190bf2-1&pdp_npi=4%40dis%21GBP%211.00%210.67%21%21%211.24%210.83%21%40210384b917052395466624455ea3d6%2112000030204924809%21sea%21GB%210%21AB&curPageLogUid=J1lVqykgnHkQ&utparam-url=scene%3Asearch%7Cquery_from%3A

Many thanks

jlecomte, 2024-01-14:
The Taloya ceiling light is used for the LGP, the background sheet and the external diffuser, so it is pretty important. If you cannot purchase the one I listed, look for other options. Here is an example that is even cheaper, from AliExpress: https://www.aliexpress.us/item/3256804883844716.html (I have not verified that it will work, you may have to make changes to the project, at which point I cannot support you)

That LED strip you linked to is 12V. You need to use a 5V LED strip (that is documented in the README, look for `Important note about the LED strip`)

jlecomte, 2024-01-14:
Addressed in [this commit](https://github.com/jlecomte/ascom-wireless-flat-panel/commit/9dee3179851d9ea506e4c751baa3d76092c932b1). Thanks!

closed 2024-01-14 by jlecomte

ktwb, 2024-08-12:
Further to this issue, what might be helpful is some sort of link or guidance to how to tell if what you've taken out of other models of these lights is an LGP. I've got a piece of something out of a light of the model you've linked but I'm not actually sure if it's an LGP or something else entirely. 

I'm not sure it is because it doesn't have the properties you've described in a light guiding plate but the construction of the light is extremely similar to the design we're using here for the LGP (LEDs flush with a large disc of something) so...

Wellco, 2024-08-12:
In reality the light guiding plate is not as critical as you might think. In my experience using device  with NINA the LEDs are usually near their lowest output, and because the image is wildly out of focus (the device is sitting on the end of your scope) the light is perfectly diffused for your flats.