jlecomte/ascom-wireless-flat-panel #3: Taloya alternative?

alexd106, 2024-01-14:
For those based in the UK who are unable to source the Taloya ceiling lamp, [this might be a suitable alternative](https://www.amazon.co.uk/gp/product/B09YRFC3JS/ref=ox_sc_act_title_2?smid=A1NNPUA7EKFNWH&th=1). I haven't tried this yet, but have ordered one and will report back. @jlecomte do you know the thickness of the Taloya diffuser? Thanks

Wellco, 2024-01-14:
@alexd106 Thanks Alex. That is very useful!

jlecomte, 2024-01-14:
The thickness of the external diffuser is a parameter of the 3D model. Look at the spreadsheet in the FreeCAD model.

acocalypso, 2024-03-10:
I build mine using that lamp 
https://a.aliexpress.com/_EyoMt5r

As I have no 3d printer I made a poor man's build :)
[image]

Wellco, 2024-03-10:
@acocalypso How is the LED light diffusion with your setup? Mine is not that, even to the naked eye, but when I compare with flats via other methods they seem to be fine. I think the reason is the light is so far our of focus and so very dim when NINA adjusts the intensity I just about get away with it.

acocalypso, 2024-03-10:
It's also not that even but I'm still missing the external 3mm diffuser, currently I'm using everything from the led lamp itself.
I did some tests 2 days ago and the produced flats where usable with Nina.

alexd106, 2024-03-24:
Sorry it's taken me a while to revisit this @Wellco. I have just finished putting my flat panel together and can confirm that the light I [linked to before](https://www.amazon.co.uk/gp/product/B09YRFC3JS/ref=ox_sc_act_title_2?smid=A1NNPUA7EKFNWH&th=1) works fine. 

It took quite a bit of jiggery-pokery to get the LGP to sit against the LEDs as the tolerances are spot on but managed to get there in the end.

Thanks again @jlecomte

[image]

[image]

[image]

Wellco, 2024-03-24:
> Sorry it's taken me a while to revisit this @Wellco. I have just finished putting my flat panel together and can confirm that the light I [linked to [...]
Looks very nice Alex. Have you tried it yet? It's very satisfying to use.

nunoagameiro, 2024-06-26:
First, congrats for the project. 
For those (like me) who don't understand anything about LGP , except the thikness of the plate, can you tell what are the major diferences that make suitable or not one LGP plate for the project ?

Wellco, 2024-06-26:
> First, congrats for the project. For those (like me) who don't understand anything about LGP , except the thikness of the plate, can you tell what are the major diferences that make suitable or not one LGP plate for the project ?

The LGP is a way to make the light diffuse and therefore even across the entire disk of the device. I had problems with this myself since I'm in the UK and could not buy the same donor  light that Julien used in his video.  However I'm not sure the light has to be that even or diffused, this is because the light is so far out of focus (its on the end of your telescope after all) and when I use mine with NINA the LEDs are near the minimum brightness. I compared my flats with other methods and could not see any differences.

If you haven't already tried, I would give the project a go. For me it was/is immensely rewarding and it makes the task of creating flats ridiculously easy.

nunoagameiro, 2024-07-02:
I've tried with a diferent LGP (i'm not able to find the Taloya brend in Portugal, so i had to use another one. 

I'm using it with a 102mm telescope and a QHY268 and i'm having a lot of dificulties to get flats with more than one second, that's why i asked if LGP of Taloya are diferent of the ones used in another brands.

Wellco, 2024-07-02:
> I've tried with a diferent LGP (i'm not able to find the Taloya brend in Portugal, so i had to use another one.
>
> I'm using it with a 102mm telescope and a QHY268 and i'm having a lot of dificulties [...]

Why do you need flats more than one second? I just let NINA control the process and it seems to do a fine job. I can't remember but I think my flats would be around 0.9 of a second.  

I don't believe the Taloya LGP is anything special (shoot me down if I a wrong), so any plate from a similar spec LED light fitting should do the job.  Good luck

cayalag, 2024-10-28:
For those of you in Europe looking for Taloya LED panels... after a lot of research I found a shop in Spain that supplies them in different sizes  (https://www.luzenled.com)

I have just received a 40 cm one, so I can trim it to fit a C11.

LeeNHawaii, 2025-04-19:
What type of material did u use to print the 3D parts and what was the nozzle size?

alexd106, 2025-04-19:
> What type of material did u use to print the 3D parts and what was the nozzle size?

Mine was printed in PETG with a 0.4 mm nozzle.