jlecomte/ascom-wireless-flat-panel #4: Schematics numbering

iancward54, 2024-01-26:
The schematics numbering for the nRF52840 seems to bare no relationship to the numbering on the actual board I am a little confused what goes where.

jlecomte, 2024-01-26:
I do not use the `nRF52840` chip directly. I use the [Adafruit Feather nRF52840 Express](https://www.adafruit.com/product/4062) board, which uses the `nRF52840` chip internally.

closed 2024-01-26 by jlecomte

iancward54, 2024-01-26:
I have the Adafruit Feather nRF52840 Express board but the numbering on the board is different than your drawings, you have D5 to D13 which don't exist on the board. It's rather confusing.
  

  
Ian

jlecomte, 2024-01-26:
Here is the pinout for that board:

https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/pinouts

It does have D5 -> D13 pins.

iancward54, 2024-01-29:
Thanks for the help so far.

Everything is connected, Bluetooth is working but no lights on the ring of LED’s.

JST-XH LED number 1 on your drawing to earth is 3 volts, number 2 to earth is 1.16 volts.

I am thinking FQP30N6L maybe wired incorrectly I have attached the wiring diagram for this component, D goes to the LED’s, G goes to the resistors and S to earth. The pins left to right are labelled G-D-S.

So close.

 

Ian

 

https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/pinouts