jlecomte/ascom-wireless-flat-panel #5: Maximum diameter of the flat panel...

rstjean24, 2024-03-19:
Bonjour Julien, I am planing to build myself the flat panel described here. I own a Celestron 9 1/4'' SCT and I am wondering if it's possible to adapt the panel to such a diameter. I understand there no issue for at least 151 mm (your own telescope) but is the illumination for up to 235 mm feasable ?

acocalypso, 2024-03-19:
> Bonjour Julien, I am planing to build myself the flat panel described here. I own a Celestron 9 1/4'' SCT and I am wondering if it's possible to adapt the panel to such a diameter. I understand [...]

Which that diameter I don't think that a led ring can illuminate the whole area. But this is just my personal thought

jlecomte, 2024-03-20:
The only way to find out is to try it. I think it will work.

closed 2024-03-20 by jlecomte