jlecomte/ascom-wireless-flat-panel #6: Minimum OTA outside diameter : <107.6mm not possible

fiesiker75, 2024-03-28:
Hi 

I`d really like to adapt this project for a Skywatcher 72ED (72mm outer diameter). However if the OTA diameter (ota_outside_diameter) is set below 107.6mm I get a "Recompute failed!" error for the Pad001 object. 

[image]

19:59:27  Updating geometry: Error build geometry(10): gp_Circ::SetRadius() - radius should be positive number
19:59:27  Invalid solution from LevenbergMarquardt solver.

I realise that this project is intended for larger telescopes. The issue seems to be that the OTA hole is smaller than the body and front peices. Is there a simple way to fix this issue?

Thank you in advance

jlecomte, 2024-03-29:
I think there is a fix possible, but it won't be pretty, i.e. the flat panel will be quite a bit bigger than your OTA, but the lip will fit the OTA. Let me think about it. I won't be able to get to this for a couple of months, but I'll see what I can do. Note: The minimum size is dictated by the size of the PCB.

jlecomte, 2024-03-29:
Added a workaround in [this commit](https://github.com/jlecomte/ascom-wireless-flat-panel/commit/3d156d5a81893f9c65a45382ab6d22e80d2a066a). That's the best I can do. I don't want to sink hours of work into this (free) project :)

closed 2024-03-29 by jlecomte