jlecomte/ascom-wireless-flat-panel #9: Low brightness of the panel even at maximum power.

giangitesta, 2024-12-28:
Hi @jlecomte,
I have made your beautiful project twice and everything works perfectly.
The only problem I have encountered is the low brightness of the panels even at maximum power.
With narrowband filters this implies exposures on the order of 240 seconds to get an acceptable flat. I would like to increase the brightness consistently.
I have already tried:
- varying the pwm frequency divider parameter in the firmware.
- externally powering the mosfet with external 5 vcc by putting the grounds in common but nothing
Do you have any advice for me?
Maybe vary the voltage divider that drives the mosfet?
I am fairly familiar with electronics but I would like to be sure how to proceed.
I thank you and wish you a happy holiday season.
Gianluca

jlecomte, 2024-12-31:
I cannot provide any advice because there could be at least a dozen reasons for this, ranging from bad components to improper wiring. You will have to troubleshoot on your own (consider it a fun challenge :) Best of luck!

closed 2024-12-31 by jlecomte

jhornos, 2025-02-07:
Hello,

Great project.
I have the same problem i have mount the flat pannel and i have conected with Nina correctly. But the bright is very low.
Did you found a solution?
Thanks