jlecomte/ascom-wireless-flat-panel #12: Circuit Board

rbolser, 2025-03-29:
I have a few circuit boards available if anyone is interested. They were made by JLCPCB. Cover my costs and shipping is all I ask.

LeeNHawaii, 2025-04-19:
I am interested in one of your boards.  What is the cost to include shipping to Hawaii..Zip 96822?

rbolser, 2025-04-19:
Looks like USPS is the cheapest at 10.65 for Priority Mail.

rbolser, 2025-04-19:
> I am interested in one of your boards. What is the cost to include shipping to Hawaii..Zip 96822?

looks like USPS is the cheapest at 10.65

LeeNHawaii, 2025-04-20:
Pls send me an email at kojaklazyATgmailDotCom to discuss payment and shipping.  Mahalo

Murz61, 2025-04-27:
I would be interested in one of the boards if you still have one.  mark.mruczek@yahoo.com. Shipping would be to Houston TX.

rbolser, 2025-04-28:
Send me your name and email address and we can figure out the details.

Murz61, 2025-04-28:
Hi,

Thank you for the quick reply.
Name: Mark Mruczek
Email: ***@***.***

Murz61, 2025-04-28:
This may be blocking my email so here it is.

mark_dot_mruczek_at_yahoo_dot_com

Name: Mark Mruczek